bluerobotics/ardusub #4: Reconfigure takeoff/land for dive/surface

rjehangir, 2016-01-23:
We will adjust the current Copter takeoff and landing code to instead perform automatic dive and surface.

Decisions:
- Surface detection method so we don't try to fly
- Bottom detection method so we don't try to dig
- Cross-over depths for speed changes

@jaxxzer

jaxxzer, 2016-02-24:
See https://github.com/bluerobotics/ardusub/commit/78c50b6c393fd8e05f299a7095e66d63f7478396

Surface mode can be taken care of in short order with this, replacing land mode.

Also I think we should take ap.land_complete and the land detector out. This essentially fills that roll for the ROV context.

rjehangir, 2016-05-12:
This has been completed. For any updates or performance issues with bottom/surface detection, file a new issue.

closed 2016-05-12 by rjehangir