bluerobotics/ardusub #6: Remove onboard baro from ekf solution.

jaxxzer, 2016-01-23:
The onboard baro is useless in estimating z axis position in an rov. Remove barometer altitude measurements from ekf when no external water pressure sensor is connected.

rjehangir, 2016-01-23:
We could possibly handle this by programmatically setting EK2_ALT_NOISE to a very high number when no external baro is present so that the internal one is ignored, more or less. We'd have to ensure that EK2_ALT_NOISE returns to its normal value on boot. Thoughts?

jaxxzer, 2016-01-24:
I'm doing this for now. I don't know if it will effectively remove the baro from the position estimate. I still want to find a method to remove the baro completely from the ekf.

closed 2016-07-06 by rjehangir