bluerobotics/ardusub #7: Update failsafes for ROV use

rjehangir, 2016-01-23:
The current failsafes largely do not apply to ROV applications and should be adjusted.

jaxxzer, 2016-01-24:
Can you elaborate on which ones should be removed, or adjusted? It appears that the radio failsafe for a RC reciever connected might not apply. I can imagine that some may want to use pwm rc input.

rjehangir, 2016-01-24:
Yes, I'll elaborate. Besides failsafes, this also includes "checks".

1. **Auto-disarm** checks for zero throttle and disarms the ROV if it is zero for 15 seconds. This is currently disabled. Are there any conditions where we would want to auto-disarm the ROV?

2. **pre_arm_rc_checks**. Not necessary for radio to be calibrated.

3. **pre_arm_gps_checks**. AUTO mode currently requires GPS, but we will never have it on an ROV.

I think that's all we need to change to allow the checks to be reenabled.

closed 2016-07-06 by rjehangir