bluerobotics/ardusub #24: After Arming Throttle too high

bradrisse, 2016-07-09:
#### Issue details
Currently using node-mavlink too communicate. After sending the arm command the motors start to spin. The issue is they spin at an RPM that is too high causing the BlueRov to dive to the bottom.

When using qgroundcontrol it also had the same issue. After chaning the joystick option to "Full Down Stick to Zero Throttle" the sub idled correctly. However I cannot find the mavlink command to correct this.

#### Version
latest

#### Platform
BlueRov

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

pixhawk

rjehangir, 2016-07-09:
@bradrisse Thanks for submitting the issue. Are you sending any other commands besides the arming command? Most importantly, are you sending MANUAL_CONTROL messages? I think that if you don't do this the RC channels will be initialized to weird numbers and probably do weird things.

bradrisse, 2016-07-09:
Im only sending the ARM command no MANUAL_CONTROL. Is this the command I need? --> http://mavlink.org/messages/common#MANUAL_CONTROL 

What would a proper command look like?
Do I send manual control before or after arming?

rjehangir, 2016-07-09:
Yes, that's the right one. It takes values of -1000 to 1000 for `x`, `y`, and `r`, and a value of 0-1000 for `z`. You should send 0 for `x-y-r` and 500 for `z` (zero throttle). You only have to send this message once to initialize the inputs, but this is the same message that you'll probably want to use to control the vehicle, so I would send it continuously.

The message also contains a bitmask for buttons. That's the last value with type uint16. If you're using a joystick controller, you can change that value when buttons are pressed. I can explain further if you're interested.

See ArduSub's joystick.cpp for more details on how MANUAL_CONTROL is used.

bradrisse, 2016-07-09:
I sent the #69 manual control successfully and got a response back from mavlink in the the console.

```
{command: 69, result: 3}

```

However this still results in the sub pushing to the bottom at full speed. I tried sending the manual control message before and after arming with no success. Additionally I found that if I start mavlink, turn on qgroundcontrol, then run the node-mavlink code I get the correct loiter motor speed. So qgroundcontrol is setting some parameter on load that I'm not currently doing in the code.

jaxxzer, 2016-07-09:
The system id of your computer sending mavlink messages must match the SYSID_MYGCS parameter on ardusub.

bradrisse, 2016-07-10:
Here is the current code I have http://pastebin.com/uF6FNeDz. It's currently set to 255, but I have tried 255, 253, 250, 245, 1, 0. 

## Console from Odroid after Mavproxy
``` 
odroid@odroid64:~$ startmavproxy
Using MAVLink 1.0
Using MAVLink 1.0
Connect /dev/ttyS1 source_system=255
Log Directory: 
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from /dev/ttyS1
online system 1
Mode(0)> Mode Mode(0)
APM: ArduSub V3.4-dev (e5962235)
APM: PX4: c1fd09a0 NuttX: 579e82d4
APM: Frame: ROV_BLUEROV_FRAME
APM: PX4v2 00350020 31334718 31383530
Received 585 parameters
```

jaxxzer, 2016-07-10:
what is the value of the SYSID_MYGCS parameter on the pixhawk?

bradrisse, 2016-07-10:
253

[image]

bradrisse, 2016-07-10:
The MANUAL_CONTROL command worked in preventing the high throttle on arm. The reason I was getting a "result:3" (UNAUTHORIZED) was because I was running a mavproxy on the odroid, which is in order to connect to qgroundcontrol, and it was conflicting with communication from the node-maxproxy.

Thank you for the help! :)

closed 2016-07-11 by rjehangir