bluerobotics/ardusub #56: qGround Control crashes 100% under Windows 10

marinesimulation, 2016-12-09:
#### Issue details
qGroundControl crashes during closing when running under Windows 10.  This is 100% reproducible and qGroundControl crashes 100% of the time.  To reproduce, install and run qGC on a Windows 10 computer.  Operate as normal.  Click on close (red "x" button of window).  Click on "Yes" when asked to close open connections.  At this point the application crashes.  Sometimes it opens the "save file" dialog box and then crashes when you select "Cancel" and sometimes it crashes while the dialog box is open (as in the screen shot below).  But it crashes 100% of the time.
[image]
[image]
[image]

#### Version
QGroundControl Development HEAD:23d3eb5 2016-12-07 17:38:58 -0800

#### Platform
BlueROV2 Vectored Thrust

#### Airframe type
Sub

#### Hardware type
Pixhawk, Raspberry Pi, Fathom-X, Dell i7 laptop, Windows 10

#### Logs

marinesimulation, 2016-12-09:
I may have found a work-around:

I first turned off "prompt to save flight data" in the qGC options.
Now before I close qGC I disconnect my laptop from the network first (removing power from the topside Fathom-X board).

I've tested this twice and both times qGC closed without issues.

rjehangir, 2016-12-10:
Addressed in https://github.com/mavlink/qgroundcontrol/pull/4315

closed 2016-12-10 by rjehangir