bluerobotics/ardusub #57: NO_IO_HEARTBEAT

mrdown11, 2016-12-13:
#### Issue details
_Please describe the problem, or desired feature_
Getting NO_IO_HEARTBEAT messages spammed in QGroundControl
#### Version
_What version was the issue encountered with_
QGC version: daily build 13-dec-2016, and also on v3.0.2 from october.
Ardusub fw: all december builds. November does not have this problem.
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter (On fw upload through gqc, I can not select no stack, only different copters and plane. I select a copter, and upload selecting your downloaded ardusub fw file.
[  ] Plane
[  ] Rover
[x] sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
BlueROV2 sub

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk px4

#### Logs
_Please provide a link to any relevant logs that show the issue_

jaxxzer, 2016-12-13:
Please format your sd card. If that works, we can close this.

marinesimulation, 2016-12-13:
@jaxxzer I have the same issue running ardusub stable on my BR2 with advanced electronics.  I swapped out the SD card in the Pixhawk and tried again.  At first the "no heartbeat" issue didn't occur, so I re-calibrated the sensors and test the ROV on the bench.  Then I powered it down to move to the tub to give it a water test.  When I powered it back up, turned on qGC and the ROV lights "flashed" and the "no heartbeat" error started.  Now whenever I power up the ROV I get this "no heartbeat" error.

jaxxzer, 2016-12-13:
@marinesimulation Please try formatting your SD cards as FAT32, and see if you still have the issue. Some low quality SD cards are not compatible with the Pixhawk. Please note that there are a lot of low quality clones of name-brand SD cards.

marinesimulation, 2016-12-13:
Unfortunately, I won't have an opportunity to test further for several weeks.  I'm using the SD cards that came with the Pixhawk (I have a spare Pixhawk) so I doubt quality of the card is an issue.

mrdown11, 2016-12-14:
I will try today. In the meantime, could you give some insight into why this would occur only with newer versions of the firmware? I got it running fine again with the november fw..

jaxxzer, 2016-12-14:
The error message was added to the code only 15 days ago: https://github.com/bluerobotics/ardusub/commit/bc45ab5409bd2aea6ad29da39f4d91ddd2e920e5. The problem is still there, you are just not being alerted to it with the old firmware.

marinesimulation, 2016-12-14:
Regardless of whether the error existed before or not, this error now renders the system inoperable which was not the case with earlier versions.

mrdown11, 2016-12-14:
Tried reformattig, seems to solve it! Thanks! Still interested why though..

marinesimulation, 2016-12-14:
Huh.  Didn't solve it for me.  I'll try again when I have time.

mrdown11, 2016-12-14:
I also reflashed the new fw (rc2), still not seeing it anymore..

jaxxzer, 2016-12-14:
The message indicates that there is a problem (corrupt filesystem on the SD card in this case) that needs to be fixed, this is desirable. If you would like to forgo logging dataflash files to the SD card in order to get rid of the error, you can set the LOG_BACKEND_TYPE parameter to None (0).

closed 2016-12-14 by jaxxzer

marinesimulation, 2016-12-14:
Re-formated a new SD card and put it in the Pixhawk and burned the stable version of firmware.  No more "heartbeat" error messages.

mrdown11, 2016-12-22:
Got the error again today.. it returned for some reason..