bluerobotics/ardusub #63: Depth/Pressure Sensor - Wrong depth in QGroundControl

sl13, 2017-04-21:
#### Issue details
_Please describe the problem, or desired feature_

I used the firmware from[ here ](http://firmware.ardusub.com/Sub/stable/v3.4/)together with the BlueROv2 before and everything works fine. Because I want some custom changes on the software I clone your repository (master) and compiled it under Linux with the command "make px4-v2-upload". After connecting with QGroundControl I noticed that the depth value in the head up display not change at all, when I pushed the BlueRov under water. It looks like the ArduSub software takes the values from the intern barometer of the pixhawk, not from the external one.

To check if I made a wiring failure, I also clone the branch "Sub-3.4" and compiled and flashed it the same way on the Pixhawk, like with the other software. With this firmware form the "Sub-3.4" branch I get the correct readings for the depth in QGroundControl.  
I think there is somewhere a bug in the barometer reading or selecting part of the software.

#### Version
Ardusub (branch: master) version from 21.04.17
QGroundControl Version: Development HEAD:da022f97c 2017-04-19 22:20:16 -0400

#### Platform
BlueRov2

#### Airframe type
vectored

#### Hardware type
pixhawk

jaxxzer, 2017-04-21:
Set GND_EXT_BUS to 1 for pixhawk.

sl13, 2017-04-21:
Thanks for the quick reply.
I changed the parameter GND_EXT_BUS from Disable to BUS0 but the problem still exist.
I also try to change the parameter GND_PRIMARY but I couldn't get the right depth values.
Is there something else I must modify in the parameters to get it work?

jaxxzer, 2017-04-21:
Click advanced, manual entry, enter 1
There is a problem with parameter metadata that does not show Bus1 as an option, I will fix that now thanks.

sl13, 2017-04-21:
Thanks now it works like it should.

closed 2017-04-21 by jaxxzer