bluerobotics/ardusub #78: Auto-config motor directions

jaxxzer, 2017-07-11:
Consider adding a function that will run each thruster individually and determine the correct value for the corresponding MOT_N_DIRECTION parameter. (Must be performed in water).

In order to do this, each motor must have some contribution to rotational movement, or depth movement, as we cannot measure forward/lateral movement alone without extra hardware.

Maybe a MAV_CMD to initiate?

jaxxzer, 2018-02-07:
https://github.com/bluerobotics/ardusub/commits/pr-motor-auto-config

jaxxzer, 2019-03-20:
> In order to do this, each motor must have some contribution to rotational movement, or depth movement, as we cannot measure forward/lateral movement alone without extra hardware.

It might be very well doable to look at the accelerometers to figure out straight forward/lateral configurations.

patrickelectric, 2019-03-25:
Moved to ArduPilot.

closed 2019-03-25 by patrickelectric