bluerobotics/ardusub #113: RC channels > 8 are not available via mavlink

jaxxzer, 2017-09-29:
Setting serial protocol to mavlink 2 does not change anything here.

patrickelectric, 2017-11-21:
@jaxxzer is was necessary to change the ArduSub channel 0 to SerialProtocol_MAVLink2.
But the doc say:
- `SerialProtocol_MAVLink2 = 2,                 // do not use - use MAVLink and provide instance of 1`

Any idea why this change is necessary ?
I have all SERIALX_PROTOCOLS in MAVLink2

### Checking the problem: ###
```
mavlink_status_t *status = mavlink_get_channel_status(chan);
// If mavlink1 is True, it'll be used mavlinkV1.
bool mavlink1 = (status->flags & MAVLINK_STATUS_FLAG_OUT_MAVLINK1) != 0;
```

### How to fix: ###
Change [ArduSub/system.cpp#L59](https://github.com/bluerobotics/ardusub/blob/master/ArduSub/system.cpp#L59) to use `SerialProtocol_MAVLink2`.

patrickelectric, 2017-11-21:
Add mavlink issue: mavlink/mavlink/issues/803

patrickelectric, 2018-03-09:
@jaxxzer I just tested the last version of ardusub and our QGC release, everything is working fine. :shipit:

jaxxzer, 2018-06-11:
What version of ardusub, what SERIAL settings? I am unable to get the numbers out.

patrickelectric, 2018-06-11:
Check the video here:
https://transfer.sh/NbRLz/deepin-screen-recorder_Select%20area_20180611175850.mp4
Serial settings:
[image]

You need to set mavproxy to use mavlink 2 with `--mav20`

patrickelectric, 2018-06-11:
Check here: https://github.com/bluerobotics/companion/pull/129

jaxxzer, 2018-06-12:
I'm not using mavproxy, I'm connected directly by usb. What's my problem then?

patrickelectric, 2018-06-12:
If you did configure the serial port to output mavlink2, we should talk with gus and don about it.

patrickelectric, 2019-03-25:
Moved to ArduPilot.

closed 2019-03-25 by patrickelectric