bluerobotics/ardusub #128: SIMSTATE.zacc peaks

patrickelectric, 2017-11-21:
ArduSub SITL have some peaks in zacc

[image]

jaxxzer, 2017-11-22:
Here is the source of your peaks: https://github.com/bluerobotics/ardusub/blob/master/libraries/SITL/SIM_Submarine.cpp#L69

patrickelectric, 2017-11-22:
Ty, I'll take a look tomorrow.

patrickelectric, 2017-11-22:
I have a new version that correct this issue, it uses a concept of force density (N/m³).

```cpp
// calculate rotational and linear accelerations
void Submarine::calculate_forces(const struct sitl_input &input, Vector3f &rot_accel, Vector3f &body_accel)
{
    rot_accel = Vector3f(0,0,0);

    auto calc_boyance_force = [=](){
        float below_water_level = position.z - frame_propriety.height/2;

        // above water level
        if (below_water_level < 0) {
            return 0.0f;
        }

        // completely above water level
        if (below_water_level > frame_propriety.height/2) {
            return frame_propriety.bouyancy_force;
        }

        return (below_water_level + frame_propriety.height/2)*frame_propriety.width*frame_propriety.length*frame_propriety.bouyancy_fd;
    };

    // slight positive buoyancy
    body_accel = Vector3f(0, 0, -calc_boyance_force());
```

closed 2017-11-22 by jaxxzer