bluerobotics/ardusub #147: Download of latest ArduSub V3.5 and V3.6-dev not working

Joschka1, 2018-03-21:
Hi at all! 
The given download link on [wiki](https://www.ardusub.com/resources/downloads.html#) for the latest stable and dev versions of ArduSub are not working. 
If i try to update my BlueRov with the QGroundControl, i got an following error:
`FirmwareUpgradeLog: PX4 releases github download failed "Error during download. Error: 99"`
Best, 
Joschka

jaxxzer, 2018-03-21:
This is a problem with QGC/windows, not ardusub.

Make sure that qgc has access through your firewall/antivirus.

Joschka1, 2018-03-21:
I'm using Debian and there is no firewall or antivirus. 
And the manual download links (http://firmware.us.ardupilot.org/Sub/latest/PX4/ArduSub-v2.px4) on your wiki (https://www.ardusub.com/resources/downloads.html) give only an 404 error.

jaxxzer, 2018-04-03:
How about http://firmware.ardupilot.org/Sub/latest/PX4/ArduSub-v2.px4?

patrickelectric, 2018-04-13:
@Joschka1 can you execute the AppImage in the terminal after setting this two variables ?
```
export QT_LOGGING_RULES="qt.network*.warning=true"
export QT_DEBUG_PLUGINS=1 
```
If possible, please send us the log.

Joschka1, 2018-04-16:
@patrickelectric yes, of course.

```
QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_x8858Q/Qt/plugins/platforms" ...
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_x8858Q/Qt/plugins/platforms/libqeglfs.so"
Found metadata in lib /tmp/.mount_x8858Q/Qt/plugins/platforms/libqeglfs.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "eglfs"
        ]
    },
    "className": "QEglFSIntegrationPlugin",
    "debug": false,
    "version": 329473
}

Got keys from plugin meta data ("eglfs")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_x8858Q/Qt/plugins/platforms/libqlinuxfb.so"
Found metadata in lib /tmp/.mount_x8858Q/Qt/plugins/platforms/libqlinuxfb.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "linuxfb"
        ]
    },
    "className": "QLinuxFbIntegrationPlugin",
    "debug": false,
    "version": 329473
}

Got keys from plugin meta data ("linuxfb")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_x8858Q/Qt/plugins/platforms/libqminimal.so"
Found metadata in lib /tmp/.mount_x8858Q/Qt/plugins/platforms/libqminimal.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "minimal"
        ]
    },
    "className": "QMinimalIntegrationPlugin",
    "debug": false,
    "version": 329473
}

Got keys from plugin meta data ("minimal")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_x8858Q/Qt/plugins/platforms/libqminimalegl.so"
Found metadata in lib /tmp/.mount_x8858Q/Qt/plugins/platforms/libqminimalegl.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "minimalegl"
        ]
    },
    "className": "QMinimalEglIntegrationPlugin",
    "debug": false,
    "version": 329473
}

Got keys from plugin meta data ("minimalegl")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_x8858Q/Qt/plugins/platforms/libqoffscreen.so"
Found metadata in lib /tmp/.mount_x8858Q/Qt/plugins/platforms/libqoffscreen.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
[... 740 lines cut ...]
loaded library "/tmp/.mount_x8858Q/Qt/plugins/imageformats/libqdds.so"
loaded library "/tmp/.mount_x8858Q/Qt/plugins/imageformats/libqgif.so"
loaded library "/tmp/.mount_x8858Q/Qt/plugins/imageformats/libqicns.so"
Cannot load library /tmp/.mount_x8858Q/Qt/plugins/imageformats/libqjp2.so: (libjasper.so.1: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden)
QLibraryPrivate::loadPlugin failed on "/tmp/.mount_x8858Q/Qt/plugins/imageformats/libqjp2.so" : "Cannot load library /tmp/.mount_x8858Q/Qt/plugins/imageformats/libqjp2.so: (libjasper.so.1: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden)"
loaded library "/tmp/.mount_x8858Q/Qt/plugins/imageformats/libqjpeg.so"
loaded library "/tmp/.mount_x8858Q/Qt/plugins/imageformats/libqtga.so"
loaded library "/tmp/.mount_x8858Q/Qt/plugins/imageformats/libqtiff.so"
loaded library "/tmp/.mount_x8858Q/Qt/plugins/imageformats/libqwbmp.so"
loaded library "/tmp/.mount_x8858Q/Qt/plugins/imageformats/libqwebp.so"
qml: 34.5 43
qml: _activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()
qml: ScreenTools: Screen.width: 2560 Screen.height: 1440 Screen.pixelDensity: 4.250980392156863
qml: ScreenTools: Screen.width: 2560 Screen.height: 1440 Screen.pixelDensity: 4.250980392156863
Unknown Fact "altimeter distance"
Unknown Fact "altimeter distance"
qrc:///qml/QGroundControl/Controls/MainToolBar.qml:114: TypeError: Cannot read property 'sub' of null
qml: ScreenTools: Screen.width: 2560 Screen.height: 1440 Screen.pixelDensity: 4.250980392156863
loaded library "/tmp/.mount_x8858Q/Qt/plugins/iconengines/libqsvgicon.so"
qml: ScreenTools: Screen.width: 2560 Screen.height: 1440 Screen.pixelDensity: 4.250980392156863
Orphaned log file count 0
QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_x8858Q/Qt/plugins/accessiblebridge" ...
QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_x8858Q/accessiblebridge" ...
loaded library "udev"
```

Joschka1, 2018-04-16:
> How about http://firmware.ardupilot.org/Sub/latest/PX4/ArduSub-v2.px4?
@jaxxzer this download works for me.

patrickelectric, 2018-04-16:
@Joschka1, It appears that everything is ok, what is the terminal output if you try to download the firmware after setting both variables [here](https://github.com/bluerobotics/ardusub/issues/147#issuecomment-381149570) ?

Joschka1, 2018-04-18:
@patrickelectric this is the output: 

```
Cannot load library crypto: (crypto: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden)
loaded library "crypto"
loaded library "ssl"
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot resolve ERR_free_strings
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_cleanup
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_init
qt.network.ssl: QSslSocket: cannot resolve sk_new_null
qt.network.ssl: QSslSocket: cannot resolve sk_push
qt.network.ssl: QSslSocket: cannot resolve sk_free
qt.network.ssl: QSslSocket: cannot resolve sk_num
qt.network.ssl: QSslSocket: cannot resolve sk_pop_free
qt.network.ssl: QSslSocket: cannot resolve sk_value
qt.network.ssl: QSslSocket: cannot resolve SSL_library_init
qt.network.ssl: QSslSocket: cannot resolve SSL_load_error_strings
qt.network.ssl: QSslSocket: cannot resolve SSL_get_ex_new_index
qt.network.ssl: QSslSocket: cannot resolve SSLv2_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv3_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv2_server_method
qt.network.ssl: QSslSocket: cannot resolve SSLv3_server_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_server_method
qt.network.ssl: QSslSocket: cannot resolve X509_STORE_CTX_get_chain
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_noconf
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_conf
qt.network.ssl: QSslSocket: cannot resolve SSLeay
qt.network.ssl: QSslSocket: cannot resolve SSLeay_version
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function sk_num
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
FirmwareUpgradeLog: PX4 releases github download failed "Error during download. Error: 99"

```

patrickelectric, 2018-04-18:
ty @Joschka1, we'll continue the investigation.

jaxxzer, 2018-04-18:
@Joschka1 The download links on the website have been fixed https://github.com/bluerobotics/ardusub-gitbook/issues/15

Joschka1, 2018-04-27:
@patrickelectric i tried it, but get still the same error 
```
 Cannot load library crypto: (crypto: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden)
loaded library "crypto"
loaded library "ssl"
FirmwareUpgradeLog: PX4 releases github download failed "Error during download. Error: 99"
```

patrickelectric, 2018-11-15:
This should be moved to QGC.

closed 2018-11-15 by patrickelectric